grid-x/modbus #26: TCP: response transaction id does not match request

andig, 2020-01-26:
I've started working with a Wallbe charger. After running my application for some time, the log will error. The only way to solve this (sofar) is restarting the application and hence the tcp modbus client:

    modbus: response transaction id '82' does not match request '83'
    modbus: response transaction id '83' does not match request '84'
    modbus: response transaction id '84' does not match request '85'
    modbus: response transaction id '85' does not match request '86'
    modbus: response transaction id '86' does not match request '87'
    modbus: response transaction id '87' does not match request '88'
    modbus: response transaction id '88' does not match request '89'
    modbus: response transaction id '89' does not match request '90'
    modbus: response transaction id '90' does not match request '91'
    modbus: response transaction id '91' does not match request '92'
    modbus: response transaction id '92' does not match request '93'
    modbus: response transaction id '93' does not match request '94'

Has anyone seen something like this before? I cannot rule out that the device might have a actual problem but I'm really confused what's going on here?

frzifus, 2020-01-27:
Looks like misbehavior of the opposition to me. Is it Modbus/TCP?

andig, 2020-01-27:
> Looks like misbehavior of the opposition to me. Is it Modbus/TCP?

Yes, mentioned this briefly in the issue title. Any hints what I could do? Even closing the connection would not help here?

jhedev, 2020-01-27:
IIRC we have seen this in the past with other devices. I'm not sure if just closing and reopening the connection helped. @gq0 knows probably more about it.

gq0, 2020-01-31:
Yes, we have seen this in the past mostly with SolarEdge inverters, occasionally Kostal. Have you tried to increase the `ProtocolRecoveryTimeout` https://github.com/grid-x/modbus/blob/master/tcpclient.go#L133

andig, 2020-02-02:
I was under the impression that would only react to `ErrTCPHeaderLength` but it's actually the other way round: https://github.com/grid-x/modbus/blob/master/tcpclient.go#L184. 

Closing for time being, much appreciated!

closed 2020-02-02 by andig