grid-x/modbus #45: Seems like this project gets less and less attention and love? Time for some changes?

RTTTC, 2021-12-25:
It seems like it sometimes takes years to approve pull requests or review issues here@ grid-x/modbus
I do understand that people get busy, life priorities and interests change.
What can be done to give this project a boost and new breath of life, save it from stagnation?
Maybe it would help to have more enthusiastic people that have write privileges and could get things mowing at a faster pace?
Any ideas? What do you guys think?

andig, 2021-12-25:
@RTTTC I don't feel this is appropiate

RTTTC, 2021-12-27:
Well , I am new to Github, young and unexperienced.. Too impatient perhaps? So maybe I do not see something? Or do not understand something?
In any case, my questions are not meant to hurt anyone's feelings, I just have a genuine concern for this project and other projects, that depend on this project..

Sadly many projects on Github with time are more and more overlooked or even completely abandoned, and it's natural,  simply such is human nature - as original developers move on to bigger better things in their life, and have no more time or desire to maintain their older projects anymore. 

Maybe there is something that can be done to improve the situation? I think this could be solved simply, as long as there are people that have the knowledge, desire and willingness to join in and help, to keep this going and evolving? 
Or is there no need for changes in your opinion?

gq0, 2021-12-29:
In terms of responsiveness, we can do better. We will revisit the CI pipeline and PR assignment beginning of next year.

andig, 2023-06-18:
> In terms of responsiveness, we can do better

@gq0 one PRs open since 2020, a number from 2022. Let me know if I can do anything to help since evcc.io is a popular consumer of this library and we'd be happy to help moving it forward.

andig, 2023-07-05:
I'm thinking about starting a fork. Would anyone be interested in contributing?