grid-x/modbus #48: Closing handlers and TCP connection state

StevenResiot, 2022-01-19:
Whenever I close a handler, any open TCP connection underneath stays open (and times out after idling - default 60seconds).

As my program dynamically opens connections to different slave servers, and as a customer is running a Modbus server that only allows one connection per IP(ugh, I know), I find myself stuck in having a single connection that customers do not always have a reference to. Additionally, I have customers wanting to access the server with different timeout settings so I'm kind of stuck as connection settings cannot really be changed after a connection is open.

Long story short: I need to open connections, do my thing, and disconnect right away. The only way I found so far is to set an IdleTimeout of a few milliseconds, but this is far from optimal.

How do I close an open TCP connection? Closing the handler doesn't close its open connections, and there's no Close method for the Client object.

Cheers

frzifus, 2022-01-19:
Did you test this in an extra debug/test program or can independent/higher-level program logic cause this?

I wonder because the [exposed close](https://github.com/grid-x/modbus/blob/master/tcpclient.go#L301) and [internal closeIdle](https://github.com/grid-x/modbus/blob/master/tcpclient.go#L346) method of  tcpTransport call the same  [internal close](https://github.com/grid-x/modbus/blob/master/tcpclient.go#L329) to terminate the TCP connection.

`net.Conn` close says:
```
// Close closes the connection.
// Any blocked Read or Write operations will be unblocked and return errors.
Close() error
```

StevenResiot, 2022-01-24:
I'm confident in my testing.
I create a handler, defer handler.Close(), and return modbus.NewClient(handler) after checking handler.Connect() doesn't return an error.
Now I have a closed handler, a derived connection from it which works for a minute but no way to Close() it.

frzifus, 2022-01-25:
Not sure if i understood your control flow correctly.

<details>
  <summary>Does it look like this?</summary>
  
```golang
func newModbusClient() modbus.Client {
	h := modbus.NewTCPClientHandler("127.0.0.1:502")
	if err := h.Connect(); err != nil {
		panic(err)
	}
	defer h.Close()
	return modbus.NewClient(h)
}

func main() {
	c := newModbusClient()
	if _, err := c.WriteSingleRegister(0xFF, 0xFF); err != nil {
		panic(err)
	}
}

```
</details>

would be good if you could share some code and logs if available.

frzifus, 2022-06-13:
Is this still a thing @StevenResiot? 

cc @Carelo @guelfey

guelfey, 2022-06-14:
@StevenResiot a minimal reproducing example code would indeed be very useful here, without this I don't know how to proceed here