grid-x/modbus #55: Using the TCP/Serial Converter with multiple devices

PVKonovalov, 2022-10-08:
How can I use this library if I want to receive data from a set of Modbus devices connected to the same serial interface of a TCP/Serial converter. In this case, I have one IP address and port and different SlaveIDs. But in the library, the SlaveID setting is at the handler level, but in my case it's better to set the SlaveID when I send ReadInputRegisters (at the client level).
Is it possible to change the SlaveID for each request?

andig, 2022-10-08:
Why don‘t you share the client between your devices?

PVKonovalov, 2022-10-10:
I didn't quite understand what is meant by share the client between my devices? I just want to send requests to multiple devices over a single TCP connection.

andig, 2022-10-10:
Use one gridx client

PVKonovalov, 2022-10-10:
Something like that?
`
handler := modbus.NewTCPClientHandler("127.0.0.1:5002")
	...
	err := handler.Connect()
	...
	client := modbus.NewClient(handler)
	...
	handler.SlaveID = 6
	results, err := client.ReadInputRegisters(0, 10)
	...
	handler.SlaveID = 7
	results, err := client.ReadInputRegisters(0, 10)

`

PVKonovalov, 2022-10-10:
Yes it works, thanks. It just looks a little out of line with the library's object hierarchy structure.

closed 2022-10-10 by PVKonovalov

frzifus, 2022-10-10:
@PVKonovalov to keep the usage similar, you can wrap the modbus client and extend your method signatures with by the salve id. Maybe that helps.