grid-x/modbus #56: Proposal: improve api

andig, 2022-11-06:
Currently, packers and transporters are closely coupled. This creates problems when sharing a single physical connection amongst multiple clients where each client talks to a different slave id. While packager and transporter already have distinct purposes, the respective types cannot be created individually from outside the library as the types are not public. Using `NewClient2` is cumbersome as the required packager/transporter types both need be created taking an address as parameter which is pointless for a packager.

I therefore propose to make the `*Packager` and `*Transporter` types public. Happy to create a PR.

This would help fixing race conditions experienced when parallelising the setup code in https://github.com/evcc-io/evcc.

frzifus, 2023-04-10:
Sure, but I probably won't make it this week.

andig, 2023-04-16:
@frzifus friendly ping. Could I get your review on https://github.com/grid-x/modbus/pull/70? #59 may be "cleaner" but is much more invasive. #70 is sweet and short.

frzifus, 2023-04-16:
i have put it on my to-do list. but i can't promise anything in the next 14 days - Iam sorry.