grid-x/modbus #73: GridBox Varta Batterie Storage Integration via Modbus

pfeifferch, 2023-04-15:
Hi

Da Ihr ja zumindest wisst, was Modbus ist und scheinbar auch schon mal damit zu tun hattet, wäre es toll wenn die Varta Storages mal via Modbus in die GridBoxen integriertert werden...

Mal mein Templates aus Home Assistant
(PS: Ich bin einfacher Anwender)

PPS: euer Adoptiertes Stiefschwester-Unternehmen der ehemaligen Vissman ViShare wünscht sich diese Integration übrigens auch schon seit langem.

---

Hi

Since you at least know what Modbus is and apparently have had something to do with it before, it would be great if the Varta Storages were integrated into the GridBoxes via Modbus.

Time my template from Home Assistant
(PS: I am a simple user)

PPS: By the way, your adopted stepsister company of the former Vissman ViShare has also wanted this integration for a long time.
      
---
```
# Varta Pulse Modbus TCP connection
modbus:
  - name: varta
    type: tcp
    host: 127.0.0.2
    port: 502
    retry_on_empty: true
    sensors:
      - name: "Varta battery SOC"
        unit_of_measurement: "%"
        address: 1068
        data_type: uint16
        slave: 255
        scan_interval: 1
      - name: "Varta battery state"
        address: 1065
        data_type: uint16
        slave: 255
        scan_interval: 1
      - name: "Varta battery capacity"
        unit_of_measurement: "10 Wh"
        address: 1071
        data_type: uint16
        slave: 255
        scan_interval: 30
      - name: "Varta battery active power"
        unit_of_measurement: "W"
        address: 1066
        slave: 255
        data_type: int16
        scan_interval: 1
      - name: "Varta battery energy counter"
        unit_of_measurement: "Wh"
        address: 1069
        count: 2
        state_class: total_increasing
        data_type: uint32
        swap: word
        slave: 255
        scan_interval: 30
      - name: "Varta battery grid power"
        unit_of_measurement: "W"
        address: 1078
        slave: 255
        data_type: int16
        scan_interval: 1
      - name: "Varta available AC charging power"
        unit_of_measurement: "W"
        address: 1083
        slave: 255
        data_type: uint16
        scan_interval: 1
      - name: "Varta available AC discharging power"
        unit_of_measurement: "W"
        address: 1084
        slave: 255
        data_type: uint16
        scan_interval: 1
      - name: "Varta PV-sensor power"
        unit_of_measurement: "W"
        address: 1102
        slave: 255
        data_type: uint16
        scan_interval: 1

```

andig, 2023-04-16:
Falsches Repo? Hier geht es nicht um HA?

frzifus, 2023-04-16:
Jop und die go modbus bibliothek hat auch nichts mit den gridBox Integrationen zu tun.

closed 2023-07-06 by gq0